tub-rip/event_based_optical_flow #4: Pretrained model and checkpoint file

Sean0120, 2022-10-25:
Hi,
    Thanks for your amazing work. Can you provide some pretrained models and checkpoint files and the config files for DSEC dataset?

shiba24, 2022-10-27:
Pretrained model -- sure.
DSEC -- ok, might take some time but I can share!

shiba24, 2022-11-04:
Hey @Sean0120 , this is the pretrained MVSEC model: https://drive.google.com/file/d/13m-waAt5X0C7f0JLBwb6KAApYxgXoA2J/view?usp=sharing

I haven't tested if this is the exact one of the table 4 in the paper (although I think this is), so I will check it when I have time. After checking I will update readme.

Can you check if you can load and run inference?

DachunKai, 2022-11-20:
@shiba24 Hey, shiba. Can you give some instructions about how to use this pretrained model to generate warped events or estimated flow such as Fig. 1. in the paper.

shiba24, 2023-01-23:
It's becoming a stale issue, I will close this, and update the readme with the link when I double check. Thanks!

shiba24, 2023-01-23:
@DachunKai I can provide the code for the inference using the model. But give me some time to clean up the code, so it may take some time. Btw, the Fig.1 in the paper is not output of DNNs, but from just model-based (optimization-based) method.