tub-rip/event_based_optical_flow #6: Bug report

DachunKai, 2022-11-20:
```python
(e2vsr) *[main][~/code/event_based_optical_flow]$ python main.py --config_file ./configs/mvsec_indoor_no_timeaware.yaml
Traceback (most recent call last):
  File "main.py", line 12, in <module>
    from src import data_loader, solver, utils, visualizer
  File "/home/kaidc/code/event_based_optical_flow/src/data_loader/__init__.py", line 7, in <module>
    from .base import DataLoaderBase
  File "/home/kaidc/code/event_based_optical_flow/src/data_loader/base.py", line 6, in <module>
    from .. import utils
  File "/home/kaidc/code/event_based_optical_flow/src/utils/__init__.py", line 2, in <module>
    from .flow_utils import (
  File "<fstring>", line 1
    (t0_location =)
                 ^
SyntaxError: invalid syntax
```

shiba24, 2022-11-20:
Hi @DachunKai thanks, but I'm not aware of this error, and the code is working on my environment.
Can you tell me how i can reproduce the error? Like Python version etc..

shiba24, 2022-11-28:
Closing, please reopen it if you still have this. Also please tell me how I can reproduce the error.

closed 2022-11-28 by shiba24