tub-rip/event_based_optical_flow #9: setting for dvxplore

eleboss, 2022-12-06:
Hi shiba,
Thanks for the amazing work. I have tried your code, and everything works well.
But for me, I am using dvxplore, I manage to modify the setting and data loader. But meets following error:

'''
Traceback (most recent call last):
  File "main.py", line 211, in <module>
    best_motion: np.ndarray = solv.optimize(batch)
  File "/home/eleboss/Documents/Vid2Curve-dev-main/ev_curve/src/solver/patch_contrast_pyramid.py", line 161, in optimize
    best_motion_per_scale, opt_result = self.run_scipy_over_scale(events)
  File "/home/eleboss/Documents/Vid2Curve-dev-main/ev_curve/src/utils/misc.py", line 127, in wrapper
    retval = func(*args, **kwargs)
  File "/home/eleboss/Documents/Vid2Curve-dev-main/ev_curve/src/solver/patch_contrast_pyramid.py", line 194, in run_scipy_over_scale
    opt_result = self.run_scipy(events, best_motion_per_scale)
  File "/home/eleboss/Documents/Vid2Curve-dev-main/ev_curve/src/solver/patch_contrast_pyramid.py", line 303, in run_scipy
    result = scipy_autograd.minimize(
  File "/home/eleboss/Documents/Vid2Curve-dev-main/ev_curve/src/solver/scipy_autograd/scipy_minimize.py", line 100, in minimize
    optim_res = sopt.minimize(
  File "/home/eleboss/anaconda3/lib/python3.8/site-packages/scipy/optimize/_minimize.py", line 689, in minimize
    res = _minimize_newtoncg(fun, x0, args, jac, hess, hessp, callback,
  File "/home/eleboss/anaconda3/lib/python3.8/site-packages/scipy/optimize/_optimize.py", line 1862, in _minimize_newtoncg
    sf = _prepare_scalar_function(
  File "/home/eleboss/anaconda3/lib/python3.8/site-packages/scipy/optimize/_optimize.py", line 263, in _prepare_scalar_function
    sf = ScalarFunction(fun, x0, args, grad, hess,
  File "/home/eleboss/anaconda3/lib/python3.8/site-packages/scipy/optimize/_differentiable_functions.py", line 158, in __init__
    self._update_fun()
  File "/home/eleboss/anaconda3/lib/python3.8/site-packages/scipy/optimize/_differentiable_functions.py", line 251, in _update_fun
    self._update_fun_impl()
  File "/home/eleboss/anaconda3/lib/python3.8/site-packages/scipy/optimize/_differentiable_functions.py", line 155, in update_fun
    self.f = fun_wrapped(self.x)
  File "/home/eleboss/anaconda3/lib/python3.8/site-packages/scipy/optimize/_differentiable_functions.py", line 137, in fun_wrapped
    fx = fun(np.copy(x), *args)
  File "/home/eleboss/anaconda3/lib/python3.8/site-packages/scipy/optimize/_optimize.py", line 76, in __call__
    self._compute_if_needed(x, *args)
  File "/home/eleboss/anaconda3/lib/python3.8/site-packages/scipy/optimize/_optimize.py", line 70, in _compute_if_needed
    fg = self.fun(x, *args)
  File "/home/eleboss/Documents/Vid2Curve-dev-main/ev_curve/src/solver/scipy_autograd/torch_wrapper.py", line 38, in get_value_and_grad
    loss = self._eval_func(input_var_)
  File "/home/eleboss/Documents/Vid2Curve-dev-main/ev_curve/src/solver/scipy_autograd/base_wrapper.py", line 109, in _eval_func
    loss = self.func(input_var)
  File "/home/eleboss/Documents/Vid2Curve-dev-main/ev_curve/src/solver/patch_contrast_pyramid.py", line 304, in <lambda>
    lambda x: self.objective_scipy(x, events, coarser_motion),
  File "/home/eleboss/Documents/Vid2Curve-dev-main/ev_curve/src/solver/patch_contrast_pyramid.py", line 453, in objective_scipy
    loss = self.calculate_cost(
  File "/home/eleboss/Documents/Vid2Curve-dev-main/ev_curve/src/solver/patch_contrast_base.py", line 281, in calculate_cost
    arg_cost = self.get_arg_for_cost(events, warp, motion_model, coarse_flow)
  File "/home/eleboss/Documents/Vid2Curve-dev-main/ev_curve/src/solver/patch_contrast_base.py", line 308, in get_arg_for_cost
    backward_events, backward_feat = self.warper.warp_event(
  File "/home/eleboss/Documents/Vid2Curve-dev-main/ev_curve/src/warp.py", line 188, in warp_event
    return self.warp_event_from_optical_flow(events, motion, ref_time)
  File "/home/eleboss/Documents/Vid2Curve-dev-main/ev_curve/src/warp.py", line 305, in warp_event_from_optical_flow
    warped_torch[..., 0] = event[..., 0] - dt * torch.gather(flow_flat[:, 0], 1, _ind)
RuntimeError: index 409412 is out of bounds for dimension 1 with size 307200

'''

Could you also share the setting for high resolution like DSCE or do you have any hint about this error?

shiba24, 2022-12-06:
Hi @eleboss thank you for the interest. Releasing the DSEC data loader and config file take some time because I'm busy now. If you can share your config and I can quickly check it.

This index error seems to me that it is most likely you set wrong `image_shape`. What is image height and width in the config?

eleboss, 2022-12-06:
Hi shiba,
Thanks for the timely reply. You are right about that. The height and width are set to 480 and 640 correspondingly. Now it works.

closed 2022-12-06 by shiba24