tub-rip/event_based_optical_flow #10: How do I output trace results in ros format（ros topic）

SharineLee, 2023-03-20:


SharineLee, 2023-03-20:
Whether the tracing result is synchronous or asynchronous？

shiba24, 2023-03-20:
Hi @SharineLee , your request to output ros format makes sense. Can you tell me how to you want to save the results?

SharineLee, 2023-03-21:
> Hi @SharineLee , your request to output ros format makes sense. Can you tell me how to you want to save the results?

I wish this code can be wrapped to a ros node.

If this node is called '**feature_tracker**'.

In the testing phase，

I wish feature_tracker can subscribe this topic:

- /dvs/camera_info  : sensor_msgs/CameraInfo   
- /dvs/events : dvs_msgs/EventArray     
- /dvs/image_raw : sensor_msgs/Image

And I wish feature_tracker can publish this topic:

- /feature_tracker/feature : sensor_msgs/PointCloudConstPtr （We call the messages in this topic **feature_msg** ）
  - **// feature_id：indicates which pixels in image frames are from the same feature point**
  -  int feature_id = feature_msg->channels[0].values[i];
  - **// camera_id:left_camera(or mono) = 0 , right_camera = 1**
  -  int camera_id = feature_msg->channels[1].values[i];// left_camera(or mono): 0 , right_camera: 1
  - **// （x,y,z）: Normalized coordinates (using camera parameters)**
  - double x = feature_msg->points[i].x;
  - double y = feature_msg->points[i].y;
  - double z = feature_msg->points[i].z;
  - **// （p_u, p_v）：Pixel coordinate**
  - double p_u = feature_msg->channels[2].values[i];
  -  double p_v = feature_msg->channels[3].values[i];
  - **// The motion speed of feature points on the pixel plane**
  - double velocity_x = feature_msg->channels[4].values[i];
  - double velocity_y = feature_msg->channels[5].values[i];

shiba24, 2023-03-21:
@SharineLee 
Thank you for the idea, I like it.
First of all, let me clarify this is a dense optical flow estimation method.
So if you meant by "feature_tracker" as the velocity (2-dimensional displacement speed) for some sparse points (pixels) in the image plane, the dense optical flow estimation might be "overkill" - too expensive.

Yet, if you want to try to integrate this module into your ROS node, I'm happy to help you. That was on my personal todo list but I was too busy to do that.
Also, please see my replies for your other issues if this code really matches your use-case. #11 #12 

Best

closed 2023-03-22 by SharineLee