tub-rip/event_based_optical_flow #16: Pre-trained Model

Alkaponn, 2023-06-25:
Hi, thanks for such great work! 

I tried to load the model file specified in [this](https://github.com/tub-rip/event_based_optical_flow/issues/4#issuecomment-1303296322) comment using pickle, but I got this error:

_pickle.UnpicklingError: A load persistent id instruction was encountered,
but no persistent_load function was specified.

Code:
```
import pickle

f = open("model_flow32/archive/data.pkl", "rb")
model = pickle.load(f)
```

If that file is no longer available, could you share it again?
Also, it would be great if you could explain how to load a pre-trained model in "run_inference.py".

shiba24, 2023-06-26:
Hi @Alkaponn , thanks for your interest.
I think you need to use `torch.load(path)` for this, not pickle. Also don't unarchive this by yourself (don't use data.pkl) but use it just as a single file (model_flow32) for the argument.

shiba24, 2023-07-05:
@Alkaponn I could release some additional codes of mine, but it will take some time.
If you want to implement it, I can help, so post/send me any questions.

Alkaponn, 2023-07-05:
> @Alkaponn I could release some additional codes of mine, but it will take some time. If you want to implement it, I can help, so post/send me any questions.

Thanks a lot. I'm planning to evaluate the model on a subset of the DSEC dataset. I've managed to download the pre-trained model but I think I need the DNN model class implementation.

Alkaponn, 2023-07-16:
Hi, could you upload the codes if you are available?

shiba24, 2024-08-21:
Hi @Alkaponn , sorry for being late but I released the EV-FlowNet model file.
https://github.com/tub-rip/event_based_optical_flow/blob/main/src/solver/nnmodels/ev_flownet.py
Please check it.

Training/inference scripts may still take some time, and I will keep track on #14 .

closed 2024-08-21 by shiba24