tub-rip/event_based_optical_flow #17: Custom Dataset

XinhuaZhang, 2023-07-05:
Hello, 

I wonder if this code supports training on custom dataset. If so, is the Python script that convert event data to HDF5 data available?

Thanks,

shiba24, 2023-07-05:
Hi, you first need to convert data to HDF5 because currently .raw format is not supported.
Can you check another library that I have been developing?: https://github.com/shiba24/event-vision-library/tree/main
Example (from text to hdf5): https://github.com/shiba24/event-vision-library/blob/main/examples/convert_text_to_hdf5.py

My long-term plan is to develop the library and include some algorithms there.

shiba24, 2023-07-06:
And after that you might want to write a custom data loader class for the Psee dataset. It's relatively straightforward and simple to implement it.
https://github.com/tub-rip/event_based_optical_flow/tree/main/src/data_loader
Actually I have one, but am not planning to release it anytime soon.

XinhuaZhang, 2023-07-07:
> Hi, you first need to convert data to HDF5 because currently .raw format is not supported. Can you check another library that I have been developing?: [...]

Thank you. I will try it.

closed 2023-07-07 by XinhuaZhang