tub-rip/event_based_optical_flow #20: flow output

valerdi, 2023-09-20:
Hi there, I've runned 

`python3 main.py --config_file ./configs/mvsec_indoor_no_timeaware.yaml`

without changing the original config you provide.

Seems like it is computing optical flow between ind1 and ind2 but is not iterating through the full dataset. Is there a parameter in yaml config to do it?  Also, is there a config to output the actual values of optical flow in a file?

One last question, in the paper the multi-scale approach use the information from previous batch to initialize the values of the different scales in the next batch, is this implemented?  

Thanks in advance

shiba24, 2023-09-20:
Hi @valerdi ,

1. Try running with `--eval` option.
2. Not in config file. But you can tweak in the visualization part of the solver, settig `save_flow=True`. Or, you can also save the flow in the main.py after the estimation is done.
3. Yes. This is the iteration: https://github.com/tub-rip/event_based_optical_flow/blob/main/main.py#L52

valerdi, 2023-09-26:
Thanks for the response. I'm playing around with the code, give me some days before closing this to see if I understand well the function you linked.

JulianKrismer, 2023-10-02:
Hey, 
I was trying the --eval option, but got the following error:
 
`2023-10-02 15:05:41,007 - src.feature_calculator - WARNING - Feature calculation is disabled in this source code.
2023-10-02 15:05:41,007 - __main__ - INFO - Sequential optimization
Traceback (most recent call last):
  File "main.py", line 157, in <module>
    assert loader.gt_flow_available  # evaluate with GT flow
AssertionError`

Is it only possible to run it with ground truth, or is there a way around it?

Kind regards and thank you for your time!

shiba24, 2023-10-02:
@JulianKrismer Hi. You can run without GT flow (with modifying the main.py.)
Without GT flow and evaluation, actually one does not need to follow the `dt=1,4` conventions in MVSEC, and you can run the flow estimation for arbitrary timestamps. What is your usecase? Do you have specific timestamp list that you want to estimate?

valerdi, 2023-10-04:
Hi @shiba24, regarding my questions, everything is running well, thanks a lot!

shiba24, 2023-10-04:
How about @JulianKrismer ?

JulianKrismer, 2023-10-04:
I wasn't able to test it yet, unfortunately.
I'm working on a master thesis, and am trying to see if optical flow could help me locate the logo of a spinning ball.

shiba24, 2023-10-05:
Ok, let me know if I could help you. Feel free to open another issue if necessary. Closing it as the original question is resolved.

closed 2023-10-05 by shiba24