tub-rip/event_based_optical_flow #22: Read AEDAT4 file

IsmailAM1999, 2023-11-27:
Hi, thank you very much for your work.

I tried with a sequence of MVSEC and it works fine, and I want to try with raw events from a mono-type event camera (.aedat4), is there any way to read this type of extension? 

I also want to try the approach to format source data in hdf5, can you guide me through the hierarchical structure? And what script should I use/modify to accept this type of data? 

Thanks in advance for your time and help.

Event camera : DVXplorer Mini

IsmailAM1999, 2023-12-04:
Thank you very much for your quick response. I am looking forward to your merge, and I will be pleased to test it out.

nikky4D, 2024-11-07:
Were you able to include this reader in the codebase?