tub-rip/event_based_optical_flow #24: configuration for higher resolution events

Goulustis, 2024-01-25:
Thanks for the great work!

I'm trying to run the method on some higher resolution data. But I'm getting terrible results. Do you have any suggestion as to what configuration to use?

Resolution: 720 x 1280
termination loss value: 8152.2227

Further, what are the requirement for "crop_height" and "crop_width" in the config? I'm getting errors [here](https://github.com/tub-rip/event_based_optical_flow/blob/756dfd0125ca0b9092d207e7aed982b763be1d32/src/solver/patch_contrast_pyramid.py#L325) where i is outside motion0. It'll run if I use a multiple of the values in the given config.

shiba24, 2024-01-26:
@Goulustis thank you for the questions. I do have working config and examples for the 1M event camera (720x1280), but the codes are not released in this repository yet. Which dataset are you using?

crop_height and width: the largest integer which is smaller than the original image shape and is dividable with `2 ** scale`
So in 720x1280 case, an example is
```
    scale: 5
    crop_height: 704
    crop_width: 1280
```

shiba24, 2024-03-06:
Closing it for inactivity. Let me know if you still have questions.

closed 2024-03-06 by shiba24