tub-rip/event_based_optical_flow #25: Runtime on DSEC dataset

Zizzzzzzz, 2024-02-29:
Hello, thank you for your excellent work! 

I am currently attempting to run this method on the DSEC dataset. I have tested a dataset containing a total of 957,721 events. The test results are satisfactory, but the model's runtime is approximately 11 minutes. I am unsure if this runtime is normal or if there is an issue with my operation at some step. I look forward to your response. Below is my configuration file:

```
solver:
  method: "pyramidal_patch_contrast_maximization"
  time_aware: False
  patch:
    initialize: "random"
    scale: 5
    crop_height: 480
    crop_width: 640
    filter_type: "bilinear"
  motion_model: "2d-translation"
  warp_direction: "first"
  parameters:
    - "trans_x"
    - "trans_y"
  cost: "hybrid"
  outer_padding: 0
  cost_with_weight:
    multi_focal_normalized_gradient_magnitude: 1.
    total_variation: 0.01
  iwe:
    method: "bilinear_vote"
    blur_sigma: 1

optimizer:
  n_iter: 40
  method: "Newton-CG"
  max_iter: 25
  parameters:
    trans_x:
      min: -150
      max: 150
    trans_y:
      min: -150
      max: 150
```

shiba24, 2024-03-11:
Hi @Zizzzzzzz , thank you for the question.
Firstly, the runtime is a limitation of this work. 11min is too much to me (I think on my Mac M1 it took tens of seconds), but could be, if your machine is slow?

You can use GPU. It naively switches to to GPU as written in README. Did you try it?

shiba24, 2024-04-16:
Closing the issue due to inactivity. Let me know if you need further help. Thanks!

closed 2024-04-16 by shiba24