tub-rip/event_based_optical_flow #28: Other dataloader

JamesYang110043, 2024-06-26:
Hi @shiba24 
First, thank you for your valuable work on this project. I have been working with your code and have found it extremely helpful for my research.

I am currently working on a project that involves the DSEC dataset, and I noticed that there isn't a specific **DSEC Dataloader** included in the repository. Could you kindly provide a DSEC Dataloade? Any help or direction would be greatly appreciated.

Thank you very much for your time and assistance.

JamesYang110043, 2024-06-26:
Could you provide a "Other solver (especially DNN)" ? 
Your assistance would be immensely helpful. Thank you.

shiba24, 2024-08-21:
@JamesYang110043 Hi, thanks for the issue. 
First let me focus on the DSEC dataloader.
It's not difficult for you to write it, though. The minimum you need to implement is `load_event`, `time_to_index`, `index_to_time`, and `get_sequence`.

akawincent, 2025-02-27:
@shiba24 Hi, Have you completed the dataloader code for DSEC? I am also curious to know how your code performs evaluation on DSEC. Thank you for your assistance！

Pinocchio-Jia, 2025-03-05:
Can you provide me with DavisDataLoader? I hope to run this project on data captured by Davis.

shiba24, 2025-03-07:
I receive many requests via github issues as well as emails. Thank you for all your interest and requests.
Will try to publish some of my dataloaders in a few months.

taehun-ryu, 2025-04-30:
Thank you for planning to release the dataloader. I sincerely hope it becomes available soon, as I’m eager to reproduce this research using additional datasets (e.g., DSEC, my custom dataset from DAVIS346).

shiba24, 2025-05-02:
ok, I will first provide Davis loader. Please wait by around June (CVPR).