tub-rip/event_based_optical_flow #29: Request for Model Code and Training Scripts

HEERAJ24, 2024-08-21:
Dear Team,

I hope this message finds you well. My name is Heeraj, and I am currently pursuing my master’s degree in Robotics at TU Dortmund. As part of my thesis, I am investigating event-based optical flow, and I recently came across your work in this repository.

The pretrained model you have provided is highly valuable and aligns closely with my research. I am working with data from a stereo setup of two event cameras, along with ground truth information of the tracked objects. In addition to optical flow, I am particularly interested in extending my thesis to include motion segmentation from optical flow.

To explore this further, I would greatly appreciate it if you could share the model code, training scripts, or any other resources that could assist me in reproducing and potentially extending your work. Access to these resources would significantly enhance my ability to experiment and contribute to the field.

Additionally, I am more than willing to contribute back to the project by improving code, adding documentation, or sharing insights from my research that may benefit the community.

Thank you for considering my request. I look forward to your response and am eager to collaborate if the opportunity arises.

Best regards,
Heeraj Ayyappan
heeraj333@gmail.com

shiba24, 2024-08-21:
Hi @HEERAJ24 , thank you for your interest in our work.
I'm very busy recently but try as much to support you. But let me clarify your requests.

The optical flow estimation based on non-DNN (optimization) was not enough for your dataset? What is the reason that you want to train DNN?
I think the model definition file is not hard to publish but the training/inference script needs lots of other code in my codebase and not sure if I can publish it in a timely manner. Meanwhile if you could provide the result from the optimization then I may be able to help you improve the results.

shiba24, 2024-08-21:
The model class is now released, you can check:
https://github.com/tub-rip/event_based_optical_flow/blob/main/src/solver/nnmodels/ev_flownet.py

shiba24, 2024-09-18:
Closing due to inactivity. Please let me know if you need any help.

closed 2024-09-18 by shiba24