tub-rip/event_based_optical_flow #32: The format of outputs

yuchenweng-private, 2024-11-21:
Hi, Shiba, Thank you for your excellent work. I have tried to implement my own dataloader to try your code on my dataset, it seems already worked,  but I have outputs like '0.png', '1.png', '2.png', '3.png' that I don't know which one is iwe, which one is optical flow.
From the generated images that I guess 3 is the optical flow, but which one is iwe generated by estimated optical flow?

shiba24, 2024-11-21:
Hi Yuchen, nice to hear that you implement your own dataloader.

The 0.png etc is coming from my visualizer's naming convention.

- `visualize_one_batch_warp` is called twice, without motion and with motion here:

https://github.com/tub-rip/event_based_optical_flow/blob/main/main.py#L178-L180

- function: https://github.com/tub-rip/event_based_optical_flow/blob/main/src/solver/patch_contrast_pyramid.py#L518-L536

My visualizer incrementally name the save file. You can optionally specify the prefix using `file_prefix` args 
https://github.com/tub-rip/event_based_optical_flow/blob/main/src/visualizer.py#L158

Hope this helps. Let me know if the result do not look good. I could help digging into the flow result.

shiba24, 2024-11-24:
@yuchenweng-private please let me know if you need more help. I close this issue for now as the original quesion is answered.

closed 2024-11-24 by shiba24