tub-rip/event_based_optical_flow #35: Question about "omit_invalid_data" in MVSEC dataloader

akawincent, 2025-02-27:
Hi @shiba24 , Thanks for your amazing work!

I noticed that when evaluating the estimated optical flow on MVSEC, there is an `omit_invalid_data` function that performs the following operations:
```python
def omit_invalid_data(self, sequence_name: str):
        logger.info(f"Use only valid frames.")
        first_valid_gt_frame = 0
        last_valid_gt_frame = -1
        if "indoor_flying1" in sequence_name:
            first_valid_gt_frame = 60
            last_valid_gt_frame = 1340
        elif "indoor_flying2" in sequence_name:
            first_valid_gt_frame = 140
            last_valid_gt_frame = 1500
        elif "indoor_flying3" in sequence_name:
            first_valid_gt_frame = 100
            last_valid_gt_frame = 1711
        elif "indoor_flying4" in sequence_name:
            first_valid_gt_frame = 104
            last_valid_gt_frame = 380
        elif "outdoor_day1" in sequence_name:
            last_valid_gt_frame = 5020
        elif "outdoor_day2" in sequence_name:
            first_valid_gt_frame = 30
            # last_valid_gt_frame = 5020

        self.gt_timestamps = self.gt_timestamps[first_valid_gt_frame:last_valid_gt_frame]
        self.U_gt_all = self.U_gt_all[first_valid_gt_frame:last_valid_gt_frame]
        self.V_gt_all = self.V_gt_all[first_valid_gt_frame:last_valid_gt_frame]

        # Update event list
        first_event_index = self.time_to_index(self.gt_timestamps[0])
        last_event_index = self.time_to_index(self.gt_timestamps[-1])
        self.left_event = self.left_event[first_event_index:last_event_index]
        self.left_ts = self.left_ts[first_event_index:last_event_index]

        self.min_ts = self.left_ts.min()
        self.max_ts = self.left_ts.max()

        # Update gray frame ts
        self.left_gray_ts = self.left_gray_ts[
            (self.gt_timestamps[0] < self.left_gray_ts)
            & (self.gt_timestamps[-1] > self.left_gray_ts)
        ]

```
I would like to know how you obtained the specific values for `first_valid_gt_frame` and `last_valid_gt_frame`. 
Additionally, when evaluating on the `outdoor_night1` sequence, are all frames selected?

Looking forward to your reply!

shiba24, 2025-03-07:
Hi @akawincent ,

> I would like to know how you obtained the specific values for first_valid_gt_frame and last_valid_gt_frame

The first/last frames in `indoor-flying{1-3}` typically do not have much motion (very small motion or fluctuating noisy data) so we omit them for the evaluation speed up. But as far as I remember, even if we test using these first/last frames the flow errors/accuracies do not change much from what are reported in the paper. This is mostly the GT flow is small (and the error becomes relatively small) for these sequences.

`outdoor_day1` ... we use specific 800 frames. This is to follow all previous works' evaluation. So, you can ignore`last_valid_gt_frame` params, it's not used in the end. If you need this 800 frames indices for fair benchmarking, I can provide further details.

`indoor_flying4`, `outdoor_day2` ... tldr, you can ignore them. These sequences are not used for the evaluation. Traditionally people do not report evaluation metrics on these sequences due to some reasons, such as being used as training sequences. But of course you can use these sequences for your own trial and visualization. I tested these sequences too, and I think it more or less worked. But at some point I realized that I do not need these sequences and so stopped using/tuning these params. I will delete these lines as they are confusing.

`outdoor_night1` ... not tested, i.e., the sequence itself is not used, so they are not set up. I think outdoor_night sequence does not provide good quality of event data, and also it has lots of flickering events. This method does not work well for the outdoor_night1, as far as i remember (you can test and confirm).

akawincent, 2025-03-10:
Get it.  Thanks for your help!

TaoZ129, 2025-03-21:
> Hi [@akawincent](https://github.com/akawincent) ,
>
>
> The first/last frames in `indoor-flying{1-3}` typically do not have much motion (very small motion or fluctuating noisy data) so we omit them for [...]

Hi, @shiba24 could you please share some details about 800 frames used in evaluation of outdoor_day1? Are these frames clipped out from *data.hdf with specific indices, and did previous works use the same indicies? I'm more than thankful if you can provide some info.
Best wishes!

shiba24, 2025-03-21:
> did previous works use the same indices?

Yes. I follow EV-Flownet (2018 and 2019). Let me dig it a bit.

shiba24, 2025-03-23:
@TaoZ129 here are the 800 frames. There are multiple ways to describe the 800 frames (all definitions should match exactly).

(1) In my codebase, in the config file, please use

```
  ind1: 9910
  ind2: 10710
```

(2)  Starting from the 10007th frame, where the 0th frame starts 3s into the bag of the original dataset.
(If you use the [extract_all_bags.sh](https://github.com/daniilidis-group/EV-FlowNet/blob/cf801ef5e4b112e08bfe6eac4f0878b1f5aac2fe/data/extract_all_bags.sh#L27) script in EV-FlowNet, the 3s will already be cropped).

Please let me know if you still have any questions. Closing the issue for now.

closed 2025-03-23 by shiba24

TaoZ129, 2025-03-24:
@shiba24 Thank you so much. I've also checked papers of EV-FlowNet as you mentioned. It says 'sequences spanning 222.4s and 240.4s', which matches the indices you provide. Thanks a lot for your help.